rzeldent/esp32cam-rtsp #119: "Failed to initialize the camera!" on (old) AI Thinker ESP32-S with AF2569 camera

Creat, 2024-03-24:
I'm trying to get this to work with my (rather old, multiple years in fact) ESP32-Cams. I got a bunch of them on a banggood sale a while back (like 3-4 € each or so), and finally have a use for them. I did try once in the past to get one to work with [esp32cam-ready](https://github.com/rzeldent/esp32cam-ready) (also by @rzeldent). Back then I did get a picture, or half a picture, or one picture but like 5-10 timeouts, but never managed to get it into an usable state. Might've been just the known bad wifi from the onboard antenna though.

Now when looking through the devices, the one that most closely resembles mine is the AI thinker one, but I have a different camera (AF2569) that isn't even listed anywhere. On the superseded esp32cam-ready project there is a picture that also looks (almost) like mine, and has the correct camera module on it though ([link to picture](https://github.com/rzeldent/esp32cam-ready/blob/develop/assets/ESP32-CAM.jpg)). The etching(text on the ESP32-S is still not quite identical (mine has less logos, only the AI one). Apparently it's an "2019dp8469" chip. I can only assume I have some sort of clone or something? I can of course provide a picture, I just don't think it'll help...

So, when I flash and connect them to my network (works fine), I can access the web-UI and configuration, but apparently it can't access the camera (`Result: -1 (ESP_FAIL)`). I did try to disable the use of PSRAM as described in #71 in [this comment](https://github.com/rzeldent/esp32cam-rtsp/issues/71#issuecomment-1962880673), but it didn't change anything.

So, anything I can do to get my camera working? It seems the base ESP32-S module is working fine, and just missing parameters, drivers or proper settings to connect to and talk to the camera. Any help would be appreciated...

rzeldent, 2024-04-16:
Hi Creat,

Strange, I also have a few very old ESP32-CAMS and they do work. Are you uploading the right version? Should be AI-Thinker

Creat, 2024-04-17:
Yes, first I tried with the bare AI-Thinker upload (as I wrote above), and get nework connectivity but the `Result: -1 (ESP_FAIL)` error remains, so nothing that uses a camera image works, obviously.

Because I didn't want to change the existing config, but follow the advice in the comment on #71 (see above), I made a copy and named it "esp32cam_custom.json" to make the change from `CAMERA_FB_IN_PSRAM` to `CAMERA_FB_IN_DRAM` (and swapped the first flags line from `'-D ESP32CAM_AI_THINKER'` to `'-D ESP32CAM_CUSTOM'`). Obviously I created the corresponding entry in platformio.ini, and then uploaded that. I didn't change anything else, but still got the same result of -1 and ESP_FAIL. I tried also removing the BOARD_HAS_PSRAM define, but no change either.

I also tried a different specimen (same old model of course), same result. How can I even figure out what the problem is? Can I generate more detailed logs somehow than just a -1 of "oh well, didn't work"?

Creat, 2024-04-17:
Ok after trying one more of these I got it to at least initialize the camera (using the unmodified AI-Thinker config). It's just as unusable as my previous esp32cam-ready attempt. Sometimes a snapshot loads (once), or it loads half a jpeg but you can see it slowly build the image (over like 5s), then stop. Even refreshing the config/status page takes a while. All that with a reasonable wifi signal reported on the status page (-56 dbm seems fine), so that doesn't seem to be the problem either.

Basically I think these modules are trash, and I'll stop wasting my time (and yours) on it. Even the ones that kinda work seem just borked anyway. Guess I'll get a few newer ones and see how that goes... Thanks for your time anyway.