rzeldent/esp32cam-rtsp #120: RTSP with Synology Surveillance Station -> Disconnect

thorstensd, 2024-04-04:
Hello,

somebody else is using Synology Surveillance Station with rtsp (rtsp://192.168.xxx.xxx:554/mjpeg/1) ?
I can connect - the test is successfull.
If I start streaming, the stream will always disconnect after ~3 secounds.
Same issue with 3 ESP32Cams. Distance to my Wifi is 2 meters without any blocking elements in between (-57dbm)
Cams, Wifi and Synology has been restartet.

Does anyone have a idea?

BR, Thorsten

thorstensd, 2024-04-04:
... by the way - I dont have issues with VLC. There is a stable rtsp stream.

rzeldent, 2024-04-06:
Hi Thorsten,

First of all, using an ESP32 for a real surveillance system is not something I recommend....

What VLC does is tat it buffers a few seconds before showing the stream. This to not fail if there is some packet loss... Something like this might be the case.
Some information could be obtained by running the ESP32 attached to the computer so will output some information....

Good luck!

thorstensd, 2024-04-06:
Thank you very much for your response.
I made a Tasmota flash - now it works perfekt without disturbes.

BR, Thorsten

closed 2024-04-06 by thorstensd