rzeldent/esp32cam-rtsp #122: Problems with Streaming (RTSP + HTTP)

lynkos, 2024-04-10:
Similar to #98, except RTSP streaming doesn't work in my case. I get the following errors:
```
Connection failed
VLC could not connect to '10.0.0.134:554'.
```

and

```
Your input can't be opened
VLC is unable to open the MRL 'rtsp://10.0.0.134:554/mjpeg/1'.
Check the log for details.
```

Here's the aforementioned [debug] log:
```
main debug: VLC media player - 3.0.20 Vetinari
main debug: Copyright © 1996-2023 the VideoLAN team
main debug: revision 3.0.20-0-g6f0d0ab126
main debug: configured with /Users/d-fu/vlc-3.0/extras/package/macosx/../../../configure  '--prefix=/Users/d-fu/vlc-3.0/build-arm64/vlc_install_dir' '--enable-macosx' '--enable-merge-ffmpeg' '--enable-osx-notifications' '--enable-faad' '--enable-flac' '--enable-theora' '--enable-shout' '--enable-ncurses' '--enable-twolame' '--enable-realrtsp' '--enable-libass' '--enable-macosx-avfoundation' '--disable-skins2' '--disable-xcb' '--disable-caca' '--disable-pulse' '--disable-sdl-image' '--disable-vnc' '--build=x86_64-apple-darwin19' '--host=aarch64-apple-darwin19' '--with-macosx-version-min=10.7' '--with-macosx-sdk=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk' '--with-breakpad=https://mac.crashes.videolan.org' 'build_alias=x86_64-apple-darwin19' 'host_alias=aarch64-apple-darwin19' 'CC=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang' 'CFLAGS=-g -arch arm64' 'LDFLAGS=-arch arm64' 'CXX=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++' 'CXXFLAGS=-g -arch arm64' 'OBJC=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang' 'OBJCFLAGS=-g -arch arm64'
```

On the other hand, HTTP streaming is extremely laggy (even at the lowest settings).

**ESP32**
```
Board type: esp32cam_ai_thinker
SDK Version: v4.4.6-dirty
CPU model: ESP32-D0WD-V3 rev. 3
CPU speed: 240 Mhz
CPU cores: 2
RAM size: 267.15 KB
PSRAM size: 4 MB
Flash size: 4 MB
```

**Diagnostics**
```
Uptime: 00:00:54
RTSP sessions: 0
Free heap: 150.26 KB
Max free block: 107.99 KB
```

**Network**
```
Host name: esp32-4022d8eecc98.local
Mac address: MAC_ADDR
Wifi mode: STA
Access point: WIFI_SSID
Signal strength: -55 dbm
IPv4 address: 10.0.0.134
IPv6 address: 0000:0000:0000:0000:0000:0000:0000:0000
```

**Camera**
```
Frame rate: 10 ms (100.0 f/s)
Frame size: QQVGA (160x120)
JPEG quality: 1 [1-100]
Brightness: 0 [-2,2]
Contrast: 0 [-2,2]
Saturation: 0 [-2,2]
Special effect: Normal
White balance: Auto
AWB gain: Auto
WB mode: Home
Exposure control: Auto
Auto exposure control (dsp): Enabled
Auto Exposure level: 0
Manual exposure value: 300
Gain control: Auto
AGC gain: 0
Gain ceiling: 2X
Black pixel correct: Auto
White pixel correct: Auto
Gamma correct: Enabled
Lens correction: Enabled
Horizontal mirror: Normal
Vertical flip: Normal
Downsize enable: Enabled
Color bar: Camera
```

rzeldent, 2024-04-10:
Hi Kiran,

Does the camera on the esp32-cam initialize properly?
This should be present somewhere in the serial logging from the esp.
If the camera can not be initialized, no connections are accepted for rtsp

lynkos, 2024-04-11:
> Hi Kiran,
> 
> Does the camera on the esp32-cam initialize properly? This should be present somewhere in the serial logging from the esp. If the camera can not be initialized, no connections are accepted for rtsp

Hi @rzeldent, thanks for reaching out! I'm not quite sure if it's initialized properly; HTTP streaming works perfectly, so I'm assuming it is?

rzeldent, 2024-04-11:
Yes, if HTTP streaming is working the camera is certainly initialized...

Try using some more conservative settings like a lower framerate (40ms) and a lower jpeg quality. Not sure but 100f/s is a bit too much..

FallcorpIndustries, 2024-05-02:
Did you resolve it ?

ylongwang2782, 2024-05-09:
I think I have meet the same problem~
All the setting is deafault.
The http stream http://192.168.8.106/stream works very well.
But the RTSP camera stream: rtsp://192.168.8.106:554/mjpeg/1 works bad in case that I am using the VLC to connect to the rtsp camera. The result is that VLC can connect to the RTSP camera but there is nothing but black

jussivuokko, 2024-07-21:
I think this might be good thread to comment about my learnings with esp32cam-rtsp. Hopefully this will help somebody like me who had experienced issues streaming from the device. Sorry for lenghty post.

So, my setup is that I currently have two esp32-cams. One is monitoring my backyard and another front door. For both I'm using two Wemos D1's since those are there handy and supplying needed 5V. I'm using Scrypted to feed Apple Homekit's security cams.

For the backyard one it's located very near to my wifi router and its wifi quality is excellent. The Wemos D1 (which supplies power to esp32-cam) was powered from Airport Extreme USB port which was also powering my air quality sensor (Ikea one). When I changed Wemos D1 to be powered from another device USB output (Eufy Homebase 2), the stream started working immediately (less than 1 second wait time). (With the previous setup I also had one problem so that after 30 days uptime connection to RTSP port 554 was constantly refused while all other operations were working properly before reboot. I haven't been able to reproduce this yet, but let's see if this continues with the new setup.)

For the front door one I had severe issues, the device didn't even power up properly. I used to power it via another Wemos D1 which was powered from Airport Express USB port. When I changed the power source to USB charger, I was able to access it. However, it had very poor network quality which might have been caused due to some gorilla tape and iron wire I used to mount the device. I didn't have energy to start changing the mounting, so I added one wifi extender (for some reason esp32-cam didn't connect to Aiport Express wifi even though it was set up as extending the wifi network, it always connected to my main Airport Extreme) near it and that did the trick. The network quality is not yet excellent even though the extender is very near to esp32-cam and stream doesn't start immediately and sometimes timeouts (but not that often).

What I'm planning is to setup better antenna to my front door device. The soldering of using external antenna doesn't seem so easy (https://randomnerdtutorials.com/esp32-cam-connect-external-antenna/), but maybe following this one would be worth trying: https://marksbench.com/electronics/esp32-cam-antenna-workaround/. I might be also changing power supply to be real 5V since in my setup Wemos D1 is supplying just a little more than 4.8V (but it seems to be still okay for esp32-cam).

TL;DR: Make sure that your power source gives 5V to the device. Check the wifi quality and consider using external antenna if quality could be better.

closed 2026-09-04 by github-actions[bot]