rzeldent/esp32cam-rtsp #125: Setting WIFI credentials from code

alkhachatryan, 2024-05-21:
Hi! For first I want to thank you for this brilliant project. There are few projects in a web for chip programming which made like this one. Really good, thank you.

But I have two questions:
1. Isn't it possible to set wifi credentials from code to immediatelly connect to that network and then start to steram? Like in a default CameraWebServer example from arduino ide? I mean I don't need to set AP, then manually config it to set wifi cerdentials so after it esp could connect to my network. The meaning of that - I want to use this code in 30 esp chips for builiding my DIY surviellance system and I dont want to manually set that config for 30 esps. Like I have a working code and what I do - just burn it in all chips I have.

2. Isn't it possible to speed-up somehow RTSP?  JPEG streaming works faster, I mean there is few delay between frames, while frame delays for RTSP are about a whole second.

RTSP speed up is not so critical, but wifi auto connection is.
I use Esp32 AI Thinker ordinary chip.
Thanks

rzeldent, 2024-06-14:
By default the module will have an name and behaved like an access point.
You can already connect to it (with the default password) and stream from there!

The Jpeg stream just sends the frames without any timing. RTS requires timing so this might be sent more often by editing the RTSP library...

AlejandroHuerta, 2025-03-09:
For those interested in doing this, you can add something like this.

```
iotWebConf.init();

iotWebConf.skipApStartup();
strcpy((*iotWebConf.getApPasswordParameter()).valueBuffer, "<some_password_for_the_ap>");
strcpy((*iotWebConf.getWifiParameterGroup())._wifiSsid, "<your_wifi_ssid>");
strcpy((*iotWebConf.getWifiParameterGroup())._wifiPassword, "<your_wifi_password>");```

rzeldent, 2025-03-31:
Yes, strncpy should be better...
However the value you provide is a const, there is no much risk of buffer overflows (unless you make it longer than the buffer)....

closed 2026-09-04 by github-actions[bot]