rzeldent/esp32cam-rtsp #128: How do I set WIFI SSID and Password before Firmware is uploaded?

AntDX316, 2024-05-31:
00:48:31.285 > AP password was not set.
00:48:31.285 > WiFi SSID was not set.
00:48:31.288 > Will stay in AP mode.

If I can configure it myself, I wouldn't have any issues as it should work??
How is the RTSP URL generated?

rzeldent, 2024-06-01:
Apparently you managed to get to the setup page. You have to set/change the password for the SSID otherwise it is not saved.
More information about setting the configuration can be found at: https://github.com/prampec/IotWebConf.

You can of course modify the code to hard code an access point. See https://docs.arduino.cc/tutorials/uno-wifi-rev2/uno-wifi-r2-web-server-ap-mode/

AntDX316, 2024-06-04:
> Apparently you managed to get to the setup page. You have to set/change the password for the SSID otherwise it is not saved. More information about setting the configuration can be found at: [...]

How do I do that through PlatformIO?

rzeldent, 2024-06-14:
Hi ant,

You can remove all the wifi configuration code and just use:

WiFi.begin("ap", "pw") to connect to your accesspoint.

AlejandroHuerta, 2025-03-09:
I posted how to do this in another issue https://github.com/rzeldent/esp32cam-rtsp/issues/125#issuecomment-2708588096

rzeldent, 2025-03-15:
Thanks Alejandro for helping! Think this can be closed!

closed 2026-09-04 by github-actions[bot]