rzeldent/esp32cam-rtsp #130: Adding support for elegoo ESP23-Wrover module (Smart Robot Car V4.0 with camera)

webermich, 2024-06-12:
Hello, 

I own an "ELEGOO Smart Robot Car V4.0 with camera" and tried to use this project to access the video stream from the camera module.

However I wasn't able to get the camera working, probably due to misconfiguration.

When using the supplied software from the provider, the camera works, but I am not all too happy with the video stream though.

Could you please give advice on how to create a proper configuration profile?

Error:
Failed to initialize camera. Error: 0x105. Frame size: VGA (640x480), frame rate: 200 ms, jpeg quality: 14

Serial output:
[serialmonitor.txt](https://github.com/user-attachments/files/15805521/serialmonitor.txt)

Using the following config:
[esp32cam_elegoo_wrover.json](https://github.com/user-attachments/files/15805454/esp32cam_elegoo_wrover.json)

------
All infos regarding the original driver:

Link to the source code: https://drive.google.com/file/d/19IENruwaLPVMKnKpy1bk7TjwF5JThqie/view
Documentation for the cars camera: https://drive.google.com/file/d/1nFgyLV-znzmQl7CN2OcnSuMXnewOJ3hG/view

I have the "ESP32-WROVER module" with “ESP32 Dev Module” configuration.
There are extra settings to be done:
**partition scheme: Huge APP (3MB no OTA...)
PSRAMM: enabled**

From the code I got the following pins:

```
#define PWDN_GPIO_NUM    -1
#define RESET_GPIO_NUM   -1
#define XCLK_GPIO_NUM    21
#define SIOD_GPIO_NUM    26
#define SIOC_GPIO_NUM    27

#define Y9_GPIO_NUM      35
#define Y8_GPIO_NUM      34
#define Y7_GPIO_NUM      39
#define Y6_GPIO_NUM      36
#define Y5_GPIO_NUM      19
#define Y4_GPIO_NUM      18
#define Y3_GPIO_NUM       5
#define Y2_GPIO_NUM       4
#define VSYNC_GPIO_NUM   25
#define HREF_GPIO_NUM    23
#define PCLK_GPIO_NUM    22

```

rzeldent, 2024-06-14:
can you not just use the wrover definition? esp32cam_freenove_wrover_kit

Otherwise take a look at the file pins in your repository and try to match this to an existing board... Very well possible  it is already there...

webermich, 2024-06-17:
Hi @rzeldent ,

what you suggest is exactly what I tried. I used  the esp32cam_freenove_wrover_kit definition, cloned it and made the pin modifications (that I thought should work). You will see the result in the file I uploaded. [esp32cam_elegoo_wrover.json](https://github.com/user-attachments/files/15805454/esp32cam_elegoo_wrover.json)

Nevertheless, it doesn't work and I was reaching out to get advice.
Maybe I have missed out something?