rzeldent/esp32cam-rtsp #131: Cannot compile firmware.elf

zetaho2003, 2024-06-23:
Processing esp32cam_seeed_xiao_esp32s3_sense (board: esp32cam_seeed_xiao_esp32s3_sense; platform: espressif32; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam_seeed_xiao_esp32s3_sense.html
PLATFORM: Espressif 32 (6.7.0) > Seeed Studio XIAO ESP32S3 Sense
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20016.0 (2.0.16)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IotWebConf @ 3.2.1
|-- Micro-RTSP @ 0.1.6
|-- micro-moustache @ 1.0.1
|-- ESPmDNS @ 2.0.0
|-- RTSPServer @ 1.0.0
Building in release mode
Linking .pio\build\esp32cam_seeed_xiao_esp32s3_sense\firmware.elf
c:/users/zetah/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot open map file C:/Users/zetah/OneDrive/桌面/esp32cam-rtsp-develop/.pio/build/esp32cam_seeed_xiao_esp32s3_sense/firmware.map: No such file or directory  
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32cam_seeed_xiao_esp32s3_sense\firmware.elf] Error 1
============================================================= [FAILED] Took 7.91 seconds =============================================================

Environment                            Status    Duration
-------------------------------------  --------  ------------
esp32cam_ai_thinker                    SUCCESS   00:00:20.836
esp32cam_espressif_esp_eye             SUCCESS   00:00:07.503
esp32cam_espressif_esp32s2_cam_board   FAILED    00:00:07.911
esp32cam_espressif_esp32s2_cam_header  FAILED    00:00:07.373
esp32cam_espressif_esp32s3_cam_lcd     FAILED    00:00:07.808
esp32cam_espressif_esp32s3_eye         FAILED    00:00:07.926
esp32cam_freenove_wrover_kit           SUCCESS   00:00:07.423
esp32cam_m5stack_camera_psram          SUCCESS   00:00:07.445
esp32cam_m5stack_camera                SUCCESS   00:00:07.486
esp32cam_m5stack_esp32cam              SUCCESS   00:00:07.373
esp32cam_m5stack_unitcam               SUCCESS   00:00:07.382
esp32cam_m5stack_unitcams3             FAILED    00:00:07.716
esp32cam_m5stack_wide                  SUCCESS   00:00:07.503
esp32cam_seeed_xiao_esp32s3_sense      FAILED    00:00:08.022
esp32cam_ttgo_t_camera                 SUCCESS   00:00:07.365
esp32cam_ttgo_t_journal                SUCCESS   00:00:07.487

What would like do i have any missing?

rzeldent, 2024-06-23:
Hi Zeta,

Heven you updated to the latest version of platformio and it's libraries?

https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_update.html#cmd-pkg-update

[PlatformIO Home](https://docs.platformio.org/en/latest/home/index.html) (GUI) → Platforms → Espressif32 → Update
or, [PlatformIO Core CLI ](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) → pio pkg update -g -p espressif3

zetaho2003, 2024-06-23:
> Hi Zeta,
>
> Heven you updated to the latest version of platformio and it's libraries?
>
> https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_update.html#cmd-pkg-update
>
> [PlatformIO [...]

Hi,

Platform Manager: espressif32@6.7.0 is already up-to-date
Tool Manager: framework-arduinoespressif32@3.20016.0 is already up-to-date
Tool Manager: tool-esptoolpy@1.40501.0 is already up-to-date
Tool Manager: tool-mkfatfs@2.0.1 is already up-to-date
Tool Manager: tool-mklittlefs@1.203.210628 is already up-to-date
Tool Manager: tool-mkspiffs@2.230.0 is already up-to-date
Tool Manager: tool-openocd-esp32@2.1100.20220706 is already up-to-date
Tool Manager: toolchain-riscv32-esp@8.4.0+2021r2-patch5 is already up-to-date
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch5 is already up-to-date
Tool Manager: toolchain-xtensa-esp32s2@8.4.0+2021r2-patch5 is already up-to-date
Tool Manager: toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5 is already up-to-date

Espressif32 is already up-to-date.

rzeldent, 2024-06-23:
This is strange because otherwise it would not pass the build: https://github.com/rzeldent/esp32cam-rtsp/actions/runs/8221705111

Will have to look at this in more detail...

rzeldent, 2024-06-25:
Hi Zeta,

I recompiled it, see the output below. Maybe this helps to find the differences...

Processing esp32cam_seeed_xiao_esp32s3_sense (board: esp32cam_seeed_xiao_esp32s3_sense; platform: espressif32; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam_seeed_xiao_esp32s3_sense.html
PLATFORM: Espressif 32 (6.7.0) > Seeed Studio XIAO ESP32S3 Sense
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20016.0 (2.0.16)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IotWebConf @ 3.2.1
|-- Micro-RTSP @ 0.1.6
|-- micro-moustache @ 1.0.1
|-- ESPmDNS @ 2.0.0
|-- RTSPServer @ 1.0.0
Building in release mode
prepare_file([".pio\build\esp32cam_seeed_xiao_esp32s3_sense\index.min.html.txt.o"], ["html\index.min.html"])
Converting .pio\build\esp32cam_seeed_xiao_esp32s3_sense\index.min.html.txt.o
revert_original_file([".pio\build\esp32cam_seeed_xiao_esp32s3_sense\index.min.html.txt.o"], ["html\index.min.html"])
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\src\main.cpp.o
Building .pio\build\esp32cam_seeed_xiao_esp32s3_sense\bootloader.bin
Generating partitions .pio\build\esp32cam_seeed_xiao_esp32s3_sense\partitions.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libba4\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libba4\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libba4\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libba4\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libba4\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libba4\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libba4\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libba4\WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libba4\WiFi\WiFiUdp.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libad0\DNSServer\DNSServer.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libfcc\FS\FS.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libfcc\FS\vfs_api.cpp.o
Archiving .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libad0\libDNSServer.a
Archiving .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libba4\libWiFi.a
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib3d8\WebServer\Parsing.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib3d8\WebServer\WebServer.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib3d8\WebServer\detail\mimetable.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib963\EEPROM\EEPROM.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib068\ESPmDNS\ESPmDNS.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib118\IotWebConf\IotWebConf.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib118\IotWebConf\IotWebConfMultipleWifi.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib118\IotWebConf\IotWebConfOptionalGroup.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib118\IotWebConf\IotWebConfParameter.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib41e\Micro-RTSP\CRtspSession.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib41e\Micro-RTSP\CStreamer.cpp.o
Archiving .pio\build\esp32cam_seeed_xiao_esp32s3_sense\libfcc\libFS.a
Archiving .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib068\libESPmDNS.a
Archiving .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib963\libEEPROM.a
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib41e\Micro-RTSP\JPEGSamples.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib41e\Micro-RTSP\OV2640.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib41e\Micro-RTSP\OV2640Streamer.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib41e\Micro-RTSP\SimStreamer.cpp.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\lib203\rtsp_server\rtsp_server.cpp.o
[... 42 lines cut ...]
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\FrameworkArduino\firmware_msc_fat.c.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32cam_seeed_xiao_esp32s3_sense\FrameworkArduino\main.cpp.o
[... cut ...]
Merged 2 ELF sections
Successfully created esp32s3 image.
=========================================================================================================================== [SUCCESS] Took 42.45 seconds ===========================================================================================================================

Environment                        Status    Duration
---------------------------------  --------  ------------
esp32cam_seeed_xiao_esp32s3_sense  SUCCESS   00:00:42.450

closed 2026-09-04 by github-actions[bot]