rzeldent/esp32cam-rtsp #132: Crash after "WIFI_READY"

Jordinator, 2024-06-27:
Hi,
maybe I missed something for my esp32cam_ai_thinker or the board is just broken. Can you please have a look into the logs that it is not the code? 

The code is crashing after the log-line "[ 37450][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY" and further below one can see that there is a csum-error.

I have used both: the sources from "main" and after that from the "devlop" branch.

13:24:12.829 > ets Jun  8 2016 00:22:57
13:24:12.829 >
13:24:12.829 > rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:24:12.829 > configsip: 0, SPIWP:0xee
13:24:12.829 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:24:12.829 > mode:DIO, clock div:2
13:24:12.829 > load:0x3fff0030,len:1184
13:24:12.829 > load:0x40078000,len:13232
13:24:12.829 > load:0x40080400,len:3028
13:24:12.829 > entry 0x400805e4
13:24:13.281 > [     4][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
13:24:13.282 > [    11][V][WebServer.cpp:88] WebServer(): WebServer::Webserver(port=80)
13:24:13.282 > [    25][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
13:24:13.736 > [   482][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
13:24:13.752 > [   517][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
13:24:13.795 > [   526][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
13:24:13.795 > [   537][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
13:24:18.789 > [  5552][I][main.cpp:332] setup(): Core debug level: 5
13:24:18.967 > [  5557][I][main.cpp:333] setup(): CPU Freq: 240 Mhz, 2 core(s)
13:24:18.967 > [  5563][I][main.cpp:334] setup(): Free heap: 250460 bytes
13:24:18.967 > [  5568][I][main.cpp:335] setup(): SDK version: v4.4.7-dirty
13:24:18.967 > [  5574][I][main.cpp:336] setup(): Board: esp32cam_ai_thinker
13:24:18.967 > [  5579][I][main.cpp:337] setup(): Starting ESP32CAM-RTSP...
13:24:18.967 > Config version: 1.6
13:24:18.967 > Config size: 333
13:24:18.967 > Wrong config version. Applying defaults.
13:24:18.967 > [iwcAll]
13:24:18.967 > |-- [iwcSys]
13:24:18.967 > |   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP-a40f99a2cff4'
13:24:18.967 > |   |-- 'iwcApPassword' with value: <hidden>
13:24:18.967 > |   |-- [iwcWifi0]
13:24:18.967 > |   |   |-- 'iwcWifiSsid' with value: ''
13:24:18.967 > |   |   \-- 'iwcWifiPassword' with value: <hidden>
13:24:18.967 > |   \-- 'iwcApTimeout' with value: '30'
13:24:18.967 > |-- [iwcCustom]
13:24:18.967 > |   \-- [camera]
13:24:18.967 > |       |-- 'fd' with value: '200'
13:24:18.967 > |       |-- 'fs' with value: 'VGA (640x480)'
13:24:18.967 > |       |-- 'q' with value: '14'
13:24:18.967 > |       |-- 'b' with value: '0'
13:24:18.967 > |       |-- 'c' with value: '0'
13:24:18.967 > |       |-- 's' with value: '0'
13:24:18.967 > |       |-- 'e' with value: 'Normal'
13:24:18.967 > |       |-- 'wb' with value: '1'
13:24:18.967 > |       |-- 'awbg' with value: '1'
13:24:18.967 > |       |-- 'wbm' with value: 'Auto'
13:24:18.967 > |       |-- 'ec' with value: '1'
13:24:18.967 > |       |-- 'aec2' with value: '1'
13:24:18.967 > |       |-- 'ael' with value: '0'
13:24:18.967 > |       |-- 'aecv' with value: '300'
13:24:18.967 > |       |-- 'gc' with value: '1'
13:24:18.967 > |       |-- 'agcg' with value: '0'
13:24:18.967 > |       |-- 'gcl' with value: '2X'
13:24:18.967 > |       |-- 'bpc' with value: '0'
13:24:18.967 > |       |-- 'wpc' with value: '1'
13:24:18.967 > |       |-- 'rg' with value: '1'
13:24:18.967 > |       |-- 'lenc' with value: '1'
13:24:18.967 > |       |-- 'hm' with value: '0'
13:24:18.967 > |       |-- 'vm' with value: '0'
13:24:18.967 > |       |-- 'dcw' with value: '1'
13:24:18.967 > |       \-- 'cb' with value: '0'
13:24:18.967 > \-- [hidden]
13:24:18.967 > [  5696][V][main.cpp:214] initialize_camera(): initialize_camera
13:24:18.967 > [  5704][I][main.cpp:216] initialize_camera(): Frame size: VGA (640x480)
13:24:18.967 > [  5711][I][main.cpp:218] initialize_camera(): JPEG quality: 14
13:24:18.967 > [  5717][I][main.cpp:220] initialize_camera(): Frame duration: 200 ms
13:24:19.020 > E (5733) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
13:24:19.020 > [  5757][E][main.cpp:388] setup(): Failed to initialize camera. Error: 0x105. Frame size: VGA (640x480), frame rate: 200 ms, jpeg quality: 14
13:24:19.545 > [  6275][V][main.cpp:214] initialize_camera(): initialize_camera
13:24:19.545 > [  6281][I][main.cpp:216] initialize_camera(): Frame size: VGA (640x480)
13:24:19.545 > [  6288][I][main.cpp:218] initialize_camera(): JPEG quality: 14
13:24:19.545 > [  6293][I][main.cpp:220] initialize_camera(): Frame duration: 200 ms
13:24:19.602 > Camera probe failed with error 0x105E (6309) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
13:24:19.602 > [  6332][E][main.cpp:388] setup(): Failed to initialize camera. Error: 0x105. Frame size: VGA (640x480), frame rate: 200 ms, jpeg quality: 14
13:24:20.124 > [  6854][V][main.cpp:214] initialize_camera(): initialize_camera
13:24:20.124 > [  6860][I][main.cpp:216] initialize_camera(): Frame size: VGA (640x480)
13:24:20.124 > [  6867][I][main.cpp:218] initialize_camera(): JPEG quality: 14
13:24:20.124 > [  6872][I][main.cpp:220] initialize_camera(): Frame duration: 200 ms
13:24:20.181 > Camera probe failed with error 0x105E (6888) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
13:24:20.181 > [  6911][E][main.cpp:388] setup(): Failed to initialize camera. Error: 0x105. Frame size: VGA (640x480), frame rate: 200 ms, jpeg quality: 14
13:24:20.686 > AP password was not set in configuration
13:24:20.687 > State changing from: 0 to 1
13:24:20.687 > [  7453][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
13:24:21.657 > ets Jun  8 2016 00:22:57
13:24:21.657 >
13:24:21.657 > rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:24:21.657 > configsip: 0, SPIWP:0xee
13:24:21.657 > clk_drv:0x00,q_drv:0x06,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x04
13:24:21.657 > mode:DIO, clock div:2
13:24:21.657 > load:0x3fff4030,len:1184
13:24:21.657 > load:0x40078000,len:13232
13:24:21.657 > load:0x40080400,len:3028
13:24:21.657 > csum err:0x2b!=0x6b
13:24:21.657 > ets_main.c 371
13:24:21.982 > ets Jun  8 2016 00:22:57

rzeldent, 2024-06-27:
Hi Jordi,

Looks like there is something not right with the camera. It is not found:
13:24:19.020 > E (5733) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)

Did you compile for the correct board or maybe attached the camera the wrong way or something like that?

Jordinator, 2024-06-28:
Hi,
thank you for your fast response!
A few weeks ago, the board and cam was working with a different code. I just ordered a new board, so we will see.

closed 2024-06-28 by Jordinator