rzeldent/esp32cam-rtsp #133: ESP32-CAM: LED Flash not working

tzahari, 2024-07-09:
Hi, nice work!
I found in issue #47 that I cloud enable the LED Flash with the 
 /flash URL. But the requests failed and I can't find it in the source code?
Also there is no hint to the URL / LED flash in the config page.

How can I enable the LED Flash?

Thanks
Best Regards Thomas

Seele0oO, 2024-07-29:
I get the same issue.
[image]

rzeldent, 2024-07-29:
Hi Thomas and Seele,

This was present in the initial version, 1.06. https://github.com/rzeldent/esp32cam-rtsp/releases/tag/1.0.6

This version only supported the ESPCAM.

This new version supports a whole range of devices, so this functionality was taken out.
It can be put in again using defines but still has to be done... In the mean time take a look at the old version...

Regards!

tzahari, 2024-07-31:
Thanks for your information

theschles, 2024-09-24:
@rzeldent wrote:
> This new version supports a whole range of devices, so this functionality was taken out. It can be put in again using defines but still has to be done... In the mean time take a look at the old version...

Yes, please add back for ESP32CAM.

Bonus points for adding an "auto-illuminate" option -- i.e. if the cam is seeing a image with a brightness below a threshold, it turns on the LED flash and adjusts the LED flash level to a point where the image meets a configured level of brightness.

vsatmydynipnet, 2024-09-24:
Good evening. Found this while searching for a solution to the led while using your rtsp server.  I put a cam in the basement and now look for a way to have the cam auto-adjusting the flash-led if is getting to dark.  This would really be a great feature for this cool software. Using ESP32-CAM (esp32cam_ai_thinker).

P7aM5qzEddT66lE, 2024-09-30:
Hi there,
THX a lot for your work!
I‘m also using the ESP32-CAM but in a 3D printer enclosure. Since it is installed in the basement I‘m also interested in lighting the LED.
Cheers!

Eric-IOT, 2025-01-02:
Hi, thank’s for this super rtsp application. I’m also very interested to be able to control the flash light for Esp32cam-AI-Thinker.

closed 2025-01-05 by rzeldent