rzeldent/esp32cam-rtsp #134: PUA.Pdf.Trojan.Embedded.JavaScript-1

baboeska, 2024-07-21:
Just did a clamav scan and this zip came back as having a trojan in the PDF, the ai_thinker_esp32cam.pdf.
Interestingly, my arduino IDE has a lot of JavaScript requests while compiling.
Strangely, I never opened that PDF.
I've noticed suspicious activity on port 9050 on phone and Linux desktop, found this while searching.
Correlation is not causation.
I hope this is of use,

Thanks

rzeldent, 2024-07-21:
Hi Banoeska,

Where is this file located? I do not see it in the repo.
But indeed watch out. Some pdf's can be infected!

baboeska, 2024-07-21:
Hi Rzeldent,
I flattened it from low orbit last night, so I sadly can't say.
Physical access and USB key upload is a possible vector re infection,and presence of esp32 security system may have contributed to infected file location.
Also noted any external or internal drive apart from main stopped functioning at software level.
I believe it may have been a gateway to a zero day exploit that uses tor socks proxy and perhaps ipv6 tunnels to communicate.
Sad to lose photos etc, but, that needed flattening.
Glad to know it's not in the repo.
Damn tir, next time virus total for sure.
I had never opened said pdf.
I'm sorry I can't be more helpful,

Best wishes

baboeska, 2024-07-21:
It showed up in the zip as well.
I wonder as to dns spoofing potential to deliver malware in place if actuals via compromised hidden proxies. Seems far fetched but it is what it is.
Have had major issues dling from GitHub without a VPN in the past.

rzeldent, 2026-08-28:
I checked it with Windows defender: no threats.
Also Virus Total Protection did not find any threats:

[image]

closed 2026-08-28 by rzeldent