rzeldent/esp32cam-rtsp #135: Compilation error

Szymon-ek, 2024-08-03:
Helo !
Setup following  instruction form wenside, new VSC , new PlatformIO and python 3.10 

during compilation continues error:

Processing esp32cam_m5stack_unitcams3 (board: esp32cam_m5stack_unitcams3; platform: espressif32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam_m5stack_unitcams3.html
PLATFORM: Espressif 32 (2024.5.13) > ESP32-CAM M5STACK UNITCAMS3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.0+sha.14a0ebd
 - tool-esptoolpy @ 4.7.3
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-riscv32-esp @ 12.2.0+20230208
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IotWebConf @ 3.2.1     
|-- Micro-RTSP @ 0.1.6     
|-- micro-moustache @ 1.0.1
|-- ESPmDNS @ 2.0.0        
|-- RTSPServer @ 1.0.0     
Building in release mode
prepare_file([".pio\build\esp32cam_m5stack_unitcams3\index.min.html.txt.o"], ["html\index.min.html"])
Converting .pio\build\esp32cam_m5stack_unitcams3\index.min.html.txt.o
revert_original_file([".pio\build\esp32cam_m5stack_unitcams3\index.min.html.txt.o"], ["html\index.min.html"])
Compiling .pio\build\esp32cam_m5stack_unitcams3\src\main.cpp.o
Building .pio\build\esp32cam_m5stack_unitcams3\bootloader.bin
Generating partitions .pio\build\esp32cam_m5stack_unitcams3\partitions.bin
esptool.py v4.7.3
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio\build\esp32cam_m5stack_unitcams3\libf98\AsyncUDP\AsyncUDP.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libaaa\Network\NetworkClient.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libaaa\Network\NetworkEvents.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libaaa\Network\NetworkInterface.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libaaa\Network\NetworkManager.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libaaa\Network\NetworkServer.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libaaa\Network\NetworkUdp.cpp.o
Archiving .pio\build\esp32cam_m5stack_unitcams3\libf98\libAsyncUDP.a
In file included from .pio/libdeps/esp32cam_m5stack_unitcams3/IotWebConf/src/IotWebConfTParameter.h:949,
                 from src/main.cpp:6:
.pio/libdeps/esp32cam_m5stack_unitcams3/IotWebConf/src/IotWebConfTParameterBuilder.h:65:42: error: expected unqualified-id before 'const'
   65 |   PrimitiveBuilder<ValueType, ParamType>(const char* id) :
      |                                          ^~~~~
.pio/libdeps/esp32cam_m5stack_unitcams3/IotWebConf/src/IotWebConfTParameterBuilder.h:65:42: error: expected ')' before 'const'
   65 |   PrimitiveBuilder<ValueType, ParamType>(const char* id) :
      |                                         ~^~~~~
      |                                          )
.pio/libdeps/esp32cam_m5stack_unitcams3/IotWebConf/src/IotWebConfTParameterBuilder.h:102:43: error: expected unqualified-id before 'const'
  102 |   Builder<IntTParameter<ValueType, base>>(const char* id) :
      |                                           ^~~~~
.pio/libdeps/esp32cam_m5stack_unitcams3/IotWebConf/src/IotWebConfTParameterBuilder.h:102:43: error: expected ')' before 'const'
  102 |   Builder<IntTParameter<ValueType, base>>(const char* id) :
      |                                          ~^~~~~
      |                                           )
.pio/libdeps/esp32cam_m5stack_unitcams3/IotWebConf/src/IotWebConfTParameterBuilder.h:111:44: error: expected unqualified-id before 'const'
  111 |   Builder<UIntTParameter<ValueType, base>>(const char* id) :
      |                                            ^~~~~
.pio/libdeps/esp32cam_m5stack_unitcams3/IotWebConf/src/IotWebConfTParameterBuilder.h:111:44: error: expected ')' before 'const'
  111 |   Builder<UIntTParameter<ValueType, base>>(const char* id) :
      |                                           ~^~~~~
[... 119 lines cut ...]
src/main.cpp:39:115: error: no matching function for call to 'iotwebconf::Builder<iotwebconf::SelectTParameter<12> >::Builder(const char [4])'
   39 | auto param_gain_ceiling = iotwebconf::Builder<iotwebconf::SelectTParameter<sizeof(camera_gain_ceilings[0])>>("gcl").label("Auto Gain ceiling").optionValues((const char *)&camera_gain_ceilings).optionNames((const char *)&camera_gain_ceilings).optionCount(sizeof(camera_gain_ceilings) / sizeof(camera_gain_ceilings[0])).nameLength(sizeof(camera_gain_ceilings[0])).defaultValue(DEFAULT_GAIN_CEILING).build();
      |                                                                                                                   ^
.pio/libdeps/esp32cam_m5stack_unitcams3/IotWebConf/src/IotWebConfTParameterBuilder.h:126:7: note: candidate: 'constexpr iotwebconf::Builder<iotwebconf::SelectTParameter<12> >::Builder(const iotwebconf::Builder<iotwebconf::SelectTParameter<12> >&)'
  126 | class Builder<SelectTParameter<len>> :
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32cam_m5stack_unitcams3/IotWebConf/src/IotWebConfTParameterBuilder.h:126:7: note:   no known conversion for argument 1 from 'const char [4]' to 'const iotwebconf::Builder<iotwebconf::SelectTParameter<12> >&'
.pio/libdeps/esp32cam_m5stack_unitcams3/IotWebConf/src/IotWebConfTParameterBuilder.h:126:7: note: candidate: 'constexpr iotwebconf::Builder<iotwebconf::SelectTParameter<12> >::Builder(iotwebconf::Builder<iotwebconf::SelectTParameter<12> >&&)'
.pio/libdeps/esp32cam_m5stack_unitcams3/IotWebConf/src/IotWebConfTParameterBuilder.h:126:7: note:   no known conversion for argument 1 from 'const char [4]' to 'iotwebconf::Builder<iotwebconf::SelectTParameter<12> >&&'
src/main.cpp: In function 'void handle_root()':
src/main.cpp:79:53: error: 'class WiFiClass' has no member named 'softAPIPv6'; did you mean 'softAPIP'?
   79 |   auto ipv6 = WiFi.getMode() == WIFI_MODE_AP ? WiFi.softAPIPv6() : WiFi.localIPv6();
      |                                                     ^~~~~~~~~~
      |                                                     softAPIP
src/main.cpp:79:73: error: 'class WiFiClass' has no member named 'localIPv6'; did you mean 'localIP'?
   79 |   auto ipv6 = WiFi.getMode() == WIFI_MODE_AP ? WiFi.softAPIPv6() : WiFi.localIPv6();
      |                                                                         ^~~~~~~~~
      |                                                                         localIP
*** [.pio\build\esp32cam_m5stack_unitcams3\src\main.cpp.o] Error 1
============================================================================================================================ [FAILED] Took 17.59 seconds ============================================================================================================================

Environment                 Status    Duration
--------------------------  --------  ------------
esp32cam_m5stack_unitcams3  FAILED    00:00:17.593
======================================================================================================================= 1 failed, 0 succeeded in 00:00:17.593 =======================================================================================================================

rzeldent, 2024-08-04:
Hi Szymon,

Looks like something is not compiling in the IotWebConf templates. I retested it and compiles without any problems. Try to update the environment to the latest version using pio update... See https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_update.html#cmd-pkg-update

Below the output of my compilation:

```txt
*  Executing task: C:\Users\rzeld\.platformio\penv\Scripts\platformio.exe run 

Processing esp32cam_m5stack_unitcams3 (board: esp32cam_m5stack_unitcams3; platform: espressif32; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam_m5stack_unitcams3.html
PLATFORM: Espressif 32 (6.7.0) > ESP32-CAM M5STACK UNITCAMS3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20016.0 (2.0.16) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IotWebConf @ 3.2.1
|-- Micro-RTSP @ 0.1.6
|-- micro-moustache @ 1.0.1
|-- ESPmDNS @ 2.0.0
|-- RTSPServer @ 1.0.0
Building in release mode
prepare_file([".pio\build\esp32cam_m5stack_unitcams3\index.min.html.txt.o"], ["html\index.min.html"])
Converting .pio\build\esp32cam_m5stack_unitcams3\index.min.html.txt.o
revert_original_file([".pio\build\esp32cam_m5stack_unitcams3\index.min.html.txt.o"], ["html\index.min.html"])
Compiling .pio\build\esp32cam_m5stack_unitcams3\src\main.cpp.o
Building .pio\build\esp32cam_m5stack_unitcams3\bootloader.bin
Generating partitions .pio\build\esp32cam_m5stack_unitcams3\partitions.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio\build\esp32cam_m5stack_unitcams3\libba4\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libba4\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libba4\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libba4\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libba4\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libba4\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libba4\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libba4\WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libba4\WiFi\WiFiUdp.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libad0\DNSServer\DNSServer.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libfcc\FS\FS.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libfcc\FS\vfs_api.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\lib3d8\WebServer\Parsing.cpp.o
Archiving .pio\build\esp32cam_m5stack_unitcams3\libba4\libWiFi.a
Archiving .pio\build\esp32cam_m5stack_unitcams3\libad0\libDNSServer.a
Compiling .pio\build\esp32cam_m5stack_unitcams3\lib3d8\WebServer\WebServer.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\lib3d8\WebServer\detail\mimetable.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\lib963\EEPROM\EEPROM.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\lib068\ESPmDNS\ESPmDNS.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libe63\IotWebConf\IotWebConf.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libe63\IotWebConf\IotWebConfMultipleWifi.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libe63\IotWebConf\IotWebConfOptionalGroup.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\libe63\IotWebConf\IotWebConfParameter.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\lib9ce\Micro-RTSP\CRtspSession.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\lib9ce\Micro-RTSP\CStreamer.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\lib9ce\Micro-RTSP\JPEGSamples.cpp.o
Archiving .pio\build\esp32cam_m5stack_unitcams3\libfcc\libFS.a
Compiling .pio\build\esp32cam_m5stack_unitcams3\lib9ce\Micro-RTSP\OV2640.cpp.o
[... 43 lines cut ...]
Compiling .pio\build\esp32cam_m5stack_unitcams3\FrameworkArduino\esp32-hal-time.c.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\FrameworkArduino\esp32-hal-tinyusb.c.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\FrameworkArduino\firmware_msc_fat.c.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\FrameworkArduino\main.cpp.o
Compiling .pio\build\esp32cam_m5stack_unitcams3\FrameworkArduino\stdlib_noniso.c.o
[... cut ...]
Retrieving maximum program size .pio\build\esp32cam_m5stack_unitcams3\firmware.elf
Checking size .pio\build\esp32cam_m5stack_unitcams3\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  25.8% (used 84412 bytes from 327680 bytes)
Flash: [=====     ]  48.8% (used 960389 bytes from 1966080 bytes)
Building .pio\build\esp32cam_m5stack_unitcams3\firmware.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
```

Szymon-ek, 2024-08-07:
Hello !

Thanks for quick relay, My log below, but the question is , can some one compile all camera models from IA thinker till ttgo and drop somewhere bin files ? Personally I am interested on Unit CamS3 but other people can use other models , same way as Tasmota have bin files in OTA section ;-)
Regards
Szymon

Result of my update:
PS C:\AA\esp32cam-rtsp> pio pkg update
Resolving esp32cam_ai_thinker dependencies...
Already up-to-date.
Resolving esp32cam_espressif_esp_eye dependencies...
Already up-to-date.
Resolving esp32cam_espressif_esp32s2_cam_board dependencies...
Resolving esp32cam_espressif_esp32s2_cam_header dependencies...
Already up-to-date.
Resolving esp32cam_espressif_esp32s3_cam_lcd dependencies...
Already up-to-date.
Resolving esp32cam_espressif_esp32s3_eye dependencies...
Already up-to-date.
Resolving esp32cam_freenove_wrover_kit dependencies...
Already up-to-date.
Resolving esp32cam_m5stack_camera_psram dependencies...
Already up-to-date.
Resolving esp32cam_m5stack_camera dependencies...
Already up-to-date.
Resolving esp32cam_m5stack_esp32cam dependencies...
Already up-to-date.
Resolving esp32cam_m5stack_unitcam dependencies...
Already up-to-date.
Resolving esp32cam_m5stack_unitcams3 dependencies...
Already up-to-date.
Resolving esp32cam_m5stack_wide dependencies...
Already up-to-date.
Resolving esp32cam_seeed_xiao_esp32s3_sense dependencies...
Already up-to-date.
Resolving esp32cam_ttgo_t_camera dependencies...
Already up-to-date.
Resolving esp32cam_ttgo_t_journal dependencies...
Already up-to-date.

rzeldent, 2024-08-08:
Hi,

They are part of a release. Take a look at: https://github.com/rzeldent/esp32cam-rtsp/releases under the Assets!

Szymon-ek, 2024-08-08:
Hello !, Yes it is but not for S3 chip ;-((((((

Unexpected chip id in image. Expected 9 but value was 0. Is this image for a different chip model?
A fatal error occurred: ESP32Cam RSTP firmware.bin is not an ESP32-S3 image. Use --force to flash anyway.

Can Some one do for ESP32 unit CAMS3 ??

Regards
Szymon

bkw, 2025-01-15:
> Looks like something is not compiling in the IotWebConf templates. I retested it and compiles without any problems. 

fwiw, I see the same errors compiling on macos, compiling on linux works fine.

tbnobody, 2025-01-23:
I encountered the same issue..... 

Changing the `platformio.ini` from 
```ini
[env]
platform = espressif32
framework = arduino
```
to
```ini
[env]
platform = espressif32@6.9.0
framework = arduino
```

fixed the problem.

rzeldent, 2025-01-25:
> [env]platform = ***@***.*** = arduino

Or keep th environment up to date..,

https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_update.html#cmd-pkg-update

This will update all the packages

bkw, 2025-01-25:
> Or keep th environment up to date..,

Sorry, I must be misunderstanding something. `pio pkg update` is what I used (results below), and it looks like @Szymon-ek had an [up-to-date environment](https://github.com/rzeldent/esp32cam-rtsp/issues/135#issuecomment-2274097964) as well.

Are you suggesting there is another way of updating that should produce working results? I've also tried `pio pkg update -p espressif32@6.9.0`, but the only thing that seems to work for me is to set `platform = espressif32@6.9.0` in platformio.ini like @tbnobody suggested above.

```console
$ uname -a
Darwin Mac 24.2.0 Darwin Kernel Version 24.2.0: Fri Dec  6 18:51:28 PST 2024; root:xnu-11215.61.5~2/RELEASE_ARM64_T8112 arm64

$ pio pkg update
Resolving esp32cam_ai_thinker dependencies...
Already up-to-date.
Resolving esp32cam_espressif_esp_eye dependencies...
Already up-to-date.
Resolving esp32cam_espressif_esp32s2_cam_board dependencies...
Already up-to-date.
Resolving esp32cam_espressif_esp32s2_cam_header dependencies...
Already up-to-date.
Resolving esp32cam_espressif_esp32s3_cam_lcd dependencies...
Already up-to-date.
Resolving esp32cam_espressif_esp32s3_eye dependencies...
Already up-to-date.
Resolving esp32cam_freenove_wrover_kit dependencies...
Already up-to-date.
Resolving esp32cam_m5stack_camera_psram dependencies...
Already up-to-date.
Resolving esp32cam_m5stack_camera dependencies...
Already up-to-date.
Resolving esp32cam_m5stack_esp32cam dependencies...
Already up-to-date.
Resolving esp32cam_m5stack_unitcam dependencies...
Already up-to-date.
Resolving esp32cam_m5stack_unitcams3 dependencies...
Already up-to-date.
Resolving esp32cam_m5stack_wide dependencies...
Already up-to-date.
Resolving esp32cam_m5stack_m5poecam_w dependencies...
Already up-to-date.
Resolving esp32cam_seeed_xiao_esp32s3_sense dependencies...
Already up-to-date.
Resolving esp32cam_ttgo_t_camera dependencies...
Already up-to-date.
Resolving esp32cam_ttgo_t_journal dependencies...
Already up-to-date.

$ pio run --environment esp32cam_ai_thinker
Processing esp32cam_ai_thinker (board: esp32cam_ai_thinker; platform: espressif32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam_ai_thinker.html
PLATFORM: Espressif 32 (51.3.6+sha.ec17486) > ESP32CAM AI Thinker
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.7 
 - framework-arduinoespressif32-libs @ 5.1.0+sha.632e0c2a9f 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IotWebConf @ 3.2.1
|-- Micro-RTSP @ 0.1.6
|-- micro-moustache @ 1.0.1
[... 311 lines cut ...]
Compiling .pio/build/esp32cam_ai_thinker/FrameworkArduino/main.cpp.o
Compiling .pio/build/esp32cam_ai_thinker/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/esp32cam_ai_thinker/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/esp32cam_ai_thinker/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp32cam_ai_thinker/libFrameworkArduino.a
Indexing .pio/build/esp32cam_ai_thinker/libFrameworkArduino.a
Linking .pio/build/esp32cam_ai_thinker/firmware.elf
Retrieving maximum program size .pio/build/esp32cam_ai_thinker/firmware.elf
Checking size .pio/build/esp32cam_ai_thinker/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  26.3% (used 86200 bytes from 327680 bytes)
Flash: [=====     ]  53.1% (used 1043037 bytes from 1966080 bytes)
Building .pio/build/esp32cam_ai_thinker/firmware.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 27 ELF sections
Successfully created esp32 image.
====================================================================== [SUCCESS] Took 7.89 seconds ======================================================================

Environment          Status    Duration
-------------------  --------  ------------
esp32cam_ai_thinker  SUCCESS   00:00:07.888
====================================================================== 1 succeeded in 00:00:07.888 ======================================================================

```

rzeldent, 2025-01-25:
Might be that there's an issue with pio pkg update. I just reinstalled platformIo and worked out of the box!

bkw, 2025-01-25:
Indeed, clearing out my ~/.platformio fixed it. Thanks.

rzeldent, 2025-02-05:
You can close it!

hn3000, 2025-03-08:
I just opened https://github.com/prampec/IotWebConf/issues/290 after I ran into the same compilation problems with IotWebConf after fully updating PlatformIO and the esp32 packages on macOS.

As for `softAPIPv6()` and `localIPv6()` -- as far as I can tell the function was renamed as part of Espressif's "[3.0.0 Network Refactoring (#8760)](https://github.com/espressif/arduino-esp32/commit/f2026f1e345970801084821d6e5f8f95a11fba47)" with the comment "* Rename IPv6 getters to clarify that they are returning LinkLocal address";
the corresponding diff is:
```diff
-    IPAddress softAPIPv6();
+    IPAddress softAPlinkLocalIPv6();
```

closed 2026-09-04 by github-actions[bot]