rzeldent/esp32cam-rtsp #136: xiao_esp32s3_sense cannot use rtsp stream

uecken, 2024-08-14:
I tried 

[image]
[image]

I changed below in platform.ini 
default_envs = esp32cam_seeed_xiao_esp32s3_sense
then upload , wifi connected.
Web Jpeg stream can be done, but RTSP stream cannot be successed.
[image]
[image]

I checked the RTSP port, then the port is not opened.
/mnt/c/Users/User$ nmap 192.168.4.1
Starting Nmap 7.80 ( https://nmap.org ) at 2024-08-14 15:28 JST
Nmap scan report for 192.168.4.1
Host is up (0.019s latency).
Not shown: 999 closed ports
PORT   STATE SERVICE
80/tcp open  http

I tried the main branch and develop branch, but same issue happend.

rzeldent, 2024-08-14:
Have you tried configuring to connect to an access point?

uecken, 2024-08-15:
How can I use Wifi station mode?
The SSID/PW setting can be shown in web GUI, but cannot ues properly.
[image]

I dont't know much about IotWebConf.

uecken, 2024-08-15:
Finally, RTSP stream connected. Thank you!!!
[image]

closed 2024-08-15 by uecken

rzeldent, 2024-08-18:
You're welcome! Nice to hear issue is resolved