rzeldent/esp32cam-rtsp #137: compilation error log

SergeKlutchenko, 2024-08-20:
Processing esp32cam_ai_thinker (board: esp32cam_ai_thinker; platform: espressif32; framework: arduino)
-------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam_ai_thinker.html
PLATFORM: Espressif 32 (4.2.0) > ESP32CAM AI Thinker
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20002.220503 (2.0.2) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 51 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IotWebConf @ 3.2.1
|-- Micro-RTSP @ 0.1.6
|-- micro-moustache @ 1.0.1
|-- ESPmDNS @ 2.0.0
|-- RTSPServer @ 1.0.0
Building in release mode
Compiling .pio/build/esp32cam_ai_thinker/src/main.cpp.o
Archiving .pio/build/esp32cam_ai_thinker/lib42a/libWiFi.a
Indexing .pio/build/esp32cam_ai_thinker/lib42a/libWiFi.a
src/main.cpp:58:72: error: call of overloaded 'String(uint64_t, int)' is ambiguous
 auto thingName = String(WIFI_SSID) + "-" + String(ESP.getEfuseMac(), 16);
                                                                        ^
In file included from /home/serge/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:159,
                 from src/main.cpp:1:
/home/serge/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:75:18: note: candidate: 'String::String(double, unsigned char)'
         explicit String(double, unsigned char decimalPlaces = 2);
                  ^~~~~~
/home/serge/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:74:18: note: candidate: 'String::String(float, unsigned char)'
         explicit String(float, unsigned char decimalPlaces = 2);
                  ^~~~~~
/home/serge/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:73:18: note: candidate: 'String::String(long unsigned int, unsigned char)'
         explicit String(unsigned long, unsigned char base = 10);
                  ^~~~~~
/home/serge/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:72:18: note: candidate: 'String::String(long int, unsigned char)'
         explicit String(long, unsigned char base = 10);
                  ^~~~~~
/home/serge/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:71:18: note: candidate: 'String::String(unsigned int, unsigned char)'
         explicit String(unsigned int, unsigned char base = 10);
                  ^~~~~~
/home/serge/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:70:18: note: candidate: 'String::String(int, unsigned char)'
         explicit String(int, unsigned char base = 10);
                  ^~~~~~
/home/serge/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:69:18: note: candidate: 'String::String(unsigned char, unsigned char)'
         explicit String(unsigned char, unsigned char base = 10);
                  ^~~~~~
src/main.cpp: In function 'esp_err_t initialize_camera()':
src/main.cpp:251:3: error: 'const camera_config_t' has no non-static data member named 'pin_sccb_sda'
   };
   ^
Archiving .pio/build/esp32cam_ai_thinker/libc69/libDNSServer.a
Indexing .pio/build/esp32cam_ai_thinker/libc69/libDNSServer.a
Compiling .pio/build/esp32cam_ai_thinker/lib1a0/FS/FS.cpp.o
*** [.pio/build/esp32cam_ai_thinker/src/main.cpp.o] Error 1
===================================== [FAILED] Took 7.24 seconds =====================================

Environment          Status    Duration
-------------------  --------  ------------
esp32cam_ai_thinker  FAILED    00:00:07.241
================================ 1 failed, 0 succeeded in 00:00:07.241 ================================
Where made mistake ?

SergeKlutchenko, 2024-08-20:
Solved  with pio pkg update

rzeldent, 2024-08-20:
Nice to hear!

closed 2024-08-20 by rzeldent