rzeldent/esp32cam-rtsp #141: Making use of the gpio ports

Kerushii, 2024-08-30:
Hi, first of all thank you for the excellent work!
I am wondering if the web server could provide API for writing gpio ports. 
Currently the stream mode will block the web server. I am thinking if the client stream could be moved to main loop instead so while it's streaming, the web server could be used to deal with gpio API requests.

rzeldent, 2024-09-01:
Hi Kerushii,

Yes this is possible but not implemented. This can be done by adding an url handler and was present in the  past.

You can use the code from version 1.0.6: https://github.com/rzeldent/esp32cam-rtsp/blob/1.0.6/src/main.cpp:

```
  // Camera flash light
  web_server.on("/flash", HTTP_GET, handle_flash);
  ```

and add a handler

Kerushii, 2024-09-01:
> Hi Kerushii,
>
> Yes this is possible but not implemented. This can be done by adding an url handler and was present in the past.
>
> You can use the code from version 1.0.6: [...]

Thank you. I think the challenge here is that according to my limited c understanding, streaming or other blocking fuunction like rtsp would make the webserver stop responding to gpio requests.

One solution I can imagine is to combine the stream and rtsp loop into the main loop, together with gpio handling logic. To do this efficiently, global objects could be used and if there are no clients occupying the webserver/rtsp, we can just return from the rtsp/stream and go to the next portion of the loop.

closed 2026-09-04 by github-actions[bot]