rzeldent/esp32cam-rtsp #146: Failed to download firmware to ai-thinker d200 board

SHILUNKA, 2024-10-24:


SHILUNKA, 2024-10-24:
I have tried to modified monitor_speed to 2000000。
PS C:\Users\Admin\Desktop\esp32cam-rtsp> pio run -t upload -e esp32cam_ai_thinker
Processing esp32cam_ai_thinker (board: esp32cam_ai_thinker; platform: espressif32; framework: arduino)
---------------------------------------------------------------------------------------------Verbose mode can be enabled via -v, --verbose option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam_ai_thinker.html
PLATFORM: Espressif 32 (6.9.0) > ESP32CAM AI Thinker
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny,
tumpa)
PACKAGES:

framework-arduinoespressif32 @ 3.20017.0 (2.0.17)
tool-esptoolpy @ 1.40501.0 (4.5.1)
tool-mkfatfs @ 2.0.1
tool-mklittlefs @ 1.203.210628 (2.3)
tool-mkspiffs @ 2.230.0 (2.30)
toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IotWebConf @ 3.2.1
|-- Micro-RTSP @ 0.1.6
|-- micro-moustache @ 1.0.1
|-- ESPmDNS @ 2.0.0
|-- RTSPServer @ 1.0.0
Building in release mode
Retrieving maximum program size .pio\build\esp32cam_ai_thinker\firmware.elf
Checking size .pio\build\esp32cam_ai_thinker\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM: [=== ] 26.3% (used 86096 bytes from 327680 bytes)
Flash: [===== ] 52.9% (used 1039477 bytes from 1966080 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM3
Uploading .pio\build\esp32cam_ai_thinker\firmware.bin
esptool.py v4.5.1
Serial port COM3
Connecting......................................
A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 2
=========================================================================================== [FAILED] Took 16.53 seconds ===========================================================================================

Environment Status Duration

rzeldent, 2024-10-24:
Hi Shilunka,

Is the IO0 grounded when rebooting and RX/TX etc connected correctly?
Please take a look at a generic ESP32 forum. There you might get some answers.

closed 2026-09-04 by github-actions[bot]