rzeldent/esp32cam-rtsp #154: panic when opening stream

33560303, 2024-11-25:
I have added a new board to this software. It is not stable so I did not do a pull request yet.
The board is functional (wifi works, I can log in and see the status). However when I open a new stream (or snapshot), the kernel panics.
Does somebody know where to look for an improvement? I have been looking into the pointer to the camera object but nothing seems wrong.

**Logging from boot to panic here, board configuration file below:**
```

16:37:56.867 > Rebooting...
16:37:57.143 > ESP-ROM:esp32s3-20210327
16:37:57.143 > Build:Mar 27 2021
16:37:57.143 > rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
16:37:57.143 > Saved PC:0x420a2a22
  #0  0x420a2a22 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_pm/pm_impl.c:855

16:37:57.143 > SPIWP:0xee
16:37:57.143 > mode:DIO, clock div:1
16:37:57.143 > load:0x3fce3808,len:0x4bc
16:37:57.143 > load:0x403c9700,len:0xbd8
16:37:57.143 > load:0x403cc700,len:0x2a0c
16:37:57.143 > entry 0x403c98d0
16:37:57.143 > [   196][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
16:37:57.143 > [   203][V][WebServer.cpp:88] WebServer(): WebServer::Webserver(port=80)
16:37:57.143 > [   229][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
16:38:02.122 > [  5253][I][main.cpp:332] setup(): Core debug level: 5
16:38:02.123 > [  5254][I][main.cpp:333] setup(): CPU Freq: 240 Mhz, 2 core(s)
16:38:02.124 > [  5255][I][main.cpp:334] setup(): Free heap: 292980 bytes
16:38:02.124 > [  5255][I][main.cpp:335] setup(): SDK version: v4.4.7-dirty
16:38:02.125 > [  5256][I][main.cpp:336] setup(): Board: esp32cam_goouuu
16:38:02.125 > [  5256][I][main.cpp:337] setup(): Starting ESP32CAM-RTSP...
16:38:02.126 > Config version: 1.6
16:38:02.126 > Config size: 333
16:38:02.128 > Loading configurations
16:38:02.128 > [iwcAll]
16:38:02.128 > |-- [iwcSys]
16:38:02.128 > |   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP'
16:38:02.128 > |   |-- 'iwcApPassword' with value: <hidden>
16:38:02.128 > |   |-- [iwcWifi0]
16:38:02.128 > |   |   |-- 'iwcWifiSsid' with value: <redacted>
16:38:02.128 > |   |   \-- 'iwcWifiPassword' with value: <hidden>
16:38:02.129 > |   \-- 'iwcApTimeout' with value: '30'
16:38:02.129 > |-- [iwcCustom]
16:38:02.129 > |   \-- [camera]
16:38:02.129 > |       |-- 'fd' with value: '200'
16:38:02.129 > |       |-- 'fs' with value: 'VGA (640x480)'
16:38:02.129 > |       |-- 'q' with value: '14'
16:38:02.129 > |       |-- 'b' with value: '0'
16:38:02.130 > |       |-- 'c' with value: '0'
16:38:02.130 > |       |-- 's' with value: '0'
16:38:02.130 > |       |-- 'e' with value: 'Normal'
16:38:02.130 > |       |-- 'wb' with value: '1'
16:38:02.130 > |       |-- 'awbg' with value: '1'
16:38:02.130 > |       |-- 'wbm' with value: 'Auto'
16:38:02.130 > |       |-- 'ec' with value: '1'
16:38:02.131 > |       |-- 'aec2' with value: '1'
16:38:02.131 > |       |-- 'ael' with value: '0'
16:38:02.131 > |       |-- 'aecv' with value: '300'
16:38:02.131 > |       |-- 'gc' with value: '1'
16:38:02.131 > |       |-- 'agcg' with value: '0'
16:38:02.131 > |       |-- 'gcl' with value: '2X'
16:38:02.132 > |       |-- 'bpc' with value: '0'
16:38:02.132 > |       |-- 'wpc' with value: '1'
16:38:02.132 > |       |-- 'rg' with value: '1'
16:38:02.132 > |       |-- 'lenc' with value: '1'
16:38:02.132 > |       |-- 'hm' with value: '0'
16:38:02.132 > |       |-- 'vm' with value: '0'
16:38:02.133 > |       |-- 'dcw' with value: '1'
16:38:02.133 > |       \-- 'cb' with value: '0'
16:38:02.133 > \-- [hidden]
16:38:02.133 > [  5264][V][main.cpp:214] initialize_camera(): initialize_camera
16:38:02.134 > [  5265][I][main.cpp:216] initialize_camera(): Frame size: VGA (640x480)
[... 126 lines cut ...]
    "wifi"
  ],
  "debug": {
    "default_tool": "esp-builtin",
    "onboard_tools": [
      "esp-builtin"
    ],
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "ESP32_CAM GOOUUU",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "",
  "vendor": "GOOUUU"
} 
```

rzeldent, 2024-11-25:
Looks like a pointer issue. Is the OV2640 instance initialized/present?

33560303, 2024-11-26:
> Looks like a pointer issue. Is the OV2640 instance initialized/present?

I think so because it says so on the HTML page.

I already thought it was a pointer issue but I cannot find the actual problem. Maybe I should buy a debugger to run trough the code because I do not have one yet.

In the meantime, do you have any idea if I can confirm/investigate the issue without a debugger?

33560303, 2024-12-24:
I have bought myself a debugger and figured it out. It turns out that the OV2640 module was broken. Replacing it solved the issue.

33560303, 2024-12-24:
I have bought myself a debugger and solved the issue. It turns out there was a hardware issue with the camera module. Replacing it solved the issue.

closed 2024-12-24 by 33560303

rzeldent, 2024-12-24:
Wow.. nice job!