rzeldent/esp32cam-rtsp #155: [NOTICE] New RTSP Server Library

rjsachse, 2025-01-04:
I was creating an RTSP server for this project [ESP32-CAM_MJPEG2SD](https://github.com/s60sc/ESP32-CAM_MJPEG2SD) (Not yet implemented) and realised it would be even more useful as a library so it can be easily integrated into other projects. The RTSP Server library enables you to stream Video, Audio, and Subtitles all in one RTSP stream. It supports UDP, TCP, and Multicast, though currently is limited to one client at a time. TCP is slower than UDP, Unicast, or Multicast, but all options are available for flexibility. This initial release was quickly put together before I returned to work, with plans for further optimization. Only tested on my own esp32s3 custom board.

Try out the example and explore the capabilities of the RTSP Server library here: [RTSPServer](https://github.com/rjsachse/RTSPServer)

rzeldent, 2025-01-05:
Hi RJachse,

Thanks for pointing to this project; it is really impressive!
Will certainly look into this!

rjsachse, 2025-01-26:
ok I have done a lot of work since, to make it easy to Intergrade. I now have multiclient support for TCP UDP and multicast and now RTSP auth so a user can set a username and password for the RTSP stream. this is all my code so if you find issues or want new features just ask.

P.S ESP32-RTSPServer latest features are now in the latest mjpeg2sd project

rzeldent, 2026-08-28:
Please take a look at the develop branch; should include audio and SRTP