rzeldent/esp32cam-rtsp #157: ESP32CAM WebServer: When I click on the facial recognition button the esp32cam resets

alerizzi2304, 2025-01-04:
### **The Problem:**
I decided to modify the html of the esp32cam web server, everything works correctly except that when I click on facial recognition, after having already clicked face detection obviously (attention: not on face detection, but on face recognition) the camera resets giving me this error, how i can fix that?

### **The Error:**
No Match Found
Guru Meditation Error: Core 1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC 0x401584e9 PS | 0x00060930 AO : 0x80104e59 A1 : 0x3ffel280
A2 : 0x3ffe133c A3 | 0x80002f80 A4 0x00002f8 A5 : 0x0000000c
A6 : 0x00000001 A7 | : 0x00000000 A8 0xbf87elc4 A9 : 0x00000000
A10 0x000000ff A11 | : 0x00000000 A12 0x0000039c A13 : 0x00000000
A14 0x00000000 A15 | : 0x00000000 SAR : 0x00000001 EXCCAUSE: 0x0000001d
EXCADDR: 0xbf87elc4 LBEG | : 0x400014fd LEND : 0x4000150d LCOUNT : Oxfffffffa
Backtrace: 0x401584e9:0x3ffe1280 0x40104e56:0x3ffe12a0 0x40104f37:0x3ffe12c0 0x40104fa3:0x3ffe1300 0x400d1909:0x3ffe1330 0x ecc.
Rebooting...
ets Jul 29 2019 12:21:46
Ist: Uxc (SW_CPU_RESET) , boot: 0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP: Oxee
clk drv: 0x00, 9_drv: 0x00, d_drv: 0x00, cs0_drv: 0800, hd_drv: 0x00, wp_drv: 0&00
mode: DIO, clock div:1
load: 0x3fff0018, len:4
load: 0x3fff001c, len: 1216
ho 0 tail 12 room 4
load: 0x40078000, len: 9720
ho 0 tail 12 room 4
load: 0x40080400, len: 6352
entry 0x400806b8

### **The Code:**
[HTMLESP32CAM.zip](https://github.com/user-attachments/files/18308433/HTMLESP32CAM.zip)

rzeldent, 2025-01-05:
Hi Alessandro,

Might be a lot of reasons. Suggest you use the esp32_exception_decoder to see where it fails.
File too large?

Good luck!

alerizzi2304, 2025-01-07:
I don't know how to do it? can you explain and help me?

rzeldent, 2025-01-25:
Hi,

I think you have the wrong repository. Esp32-cam does not have that functionality.

Normally you could use the exception decoder to pinpoint the problem.

Good luck!

closed 2026-09-04 by github-actions[bot]