rzeldent/esp32cam-rtsp #160: Camera settings ignored?

0wwafa, 2025-01-18:
Whatever I set in "FRAME SIZE", both the web streaming (fast) and RTSP stream seems not to be affected and just output 640x480.

Also, why the web streaming (jpg) is so fast and the rtsp instead is so slow (like 5 fps vs 30 or so)

rzeldent, 2025-01-18:
Hi Robert,

RTSP requires a fixed clock and this means that in the loop the frame will be outputted every xx ms.
The web streaming just grabs a frame and sends it without delay.

See handle stream:

```
  while (client.connected())
  {
    client.write("\r\n--" STREAM_CONTENT_BOUNDARY "\r\n");
    cam.run();
    client.write("Content-Type: image/jpeg\r\nContent-Length: ");
    sprintf(size_buf, "%d\r\n\r\n", cam.getSize());
    client.write(size_buf);
    client.write(cam.getfb(), cam.getSize());
  }
```

RTSP is a rather complex protocol....

0wwafa, 2025-01-19:
I understand. but why are camera settings ignored?
The camera (DC 2640 V3) can grab frames at 1600x1200 but it only seems to grab 640x480 with esp32cam-rtsp :(
http://www.uctronics.com/download/OV2640_DS.pdf

AlejandroHuerta, 2025-03-09:
This is due to the framesize parameter not being sent to the camera after being updated. You can do something like this to fix it.

```
  camera->set_framesize(camera, lookup_frame_size(param_frame_size.value()));
  camera->set_quality(camera, param_jpg_quality.value());
```

closed 2026-09-04 by github-actions[bot]