rzeldent/esp32cam-rtsp #162: Can't load RTSP stream in Home Assistant

andrewjsmith76, 2025-03-01:
Hi!

I successfully installed the code onto my AI THINKER ESP32CAM... The stream works in VLC.

But when I try to add it to Home Assistant using the "Generic Camera" integration (which works well with other RTSP streams), I get the following error message:

`Failed to start WebRTC stream: webrtc: streams: exec/rtsp [rtsp @ 0x7f80b90000] method DESCRIBE failed: 404 Not Found [in#0 @ 0x7f80e12b00] Error opening input: Server returned 404 Not Found Error opening input file rtsp://127.0.0.1:18554/camera.192_168_1_133?audio&source=ffmpeg:camera.192_168_1_133%23audio%3Dopus%23query%3Dlog_level%3Ddebug&log_level=debug. Error opening input files: Server returned 404 Not Found`

This is the output in the serial log:
```
[20:39:56]ets Jul 29 2019 12:21:46
[20:39:56]
[20:39:56]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[20:39:56]configsip: 0, SPIWP:0xee
[20:39:56]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:39:56]mode:DIO, clock div:2
[20:39:56]load:0x3fff0030,len:1184
[20:39:56]load:0x40078000,len:13232
[20:39:56]load:0x40080400,len:3028
[20:39:56]entry 0x400805e4
[20:39:56][     4][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[20:39:56][    11][V][WebServer.cpp:88] WebServer(): WebServer::Webserver(port=80)
[20:39:56][    25][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[20:39:57][   482][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[20:39:57][   514][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[20:39:57][   523][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[20:39:57][   534][V][esp32-hal-uart.c:467] uartBegin(): UART0 initialization done.
[20:40:02][  5549][I][main.cpp:332] setup(): Core debug level: 5
[20:40:02][  5554][I][main.cpp:333] setup(): CPU Freq: 240 Mhz, 2 core(s)
[20:40:02][  5560][I][main.cpp:334] setup(): Free heap: 250428 bytes
[20:40:02][  5565][I][main.cpp:335] setup(): SDK version: v4.4.7-dirty
[20:40:02][  5571][I][main.cpp:336] setup(): Board: esp32cam_ai_thinker
[20:40:02][  5576][I][main.cpp:337] setup(): Starting ESP32CAM-RTSP...
[20:40:02]Config version: 1.6
[20:40:02]Config size: 333
[20:40:02]Loading configurations
[20:40:02][iwcAll]
[20:40:02]|-- [iwcSys]
[20:40:02]|   |-- 'iwcThingName' with value: 'ESP32CAM-RTSP_AquaLightLEDWatch'
[20:40:02]|   |-- 'iwcApPassword' with value: <hidden>
[20:40:02]|   |-- [iwcWifi0]
[20:40:02]|   |   |-- 'iwcWifiSsid' with value: 'MAZnet'
[20:40:02]|   |   \-- 'iwcWifiPassword' with value: <hidden>
[20:40:02]|   \-- 'iwcApTimeout' with value: '2'
[20:40:02]|-- [iwcCustom]
[20:40:02]|   \-- [camera]
[20:40:02]|       |-- 'fd' with value: '200'
[20:40:02]|       |-- 'fs' with value: 'VGA (640x480)'
[20:40:02]|       |-- 'q' with value: '14'
[20:40:02]|       |-- 'b' with value: '0'
[20:40:02]|       |-- 'c' with value: '0'
[20:40:02]|       |-- 's' with value: '0'
[20:40:02]|       |-- 'e' with value: 'Normal'
[20:40:02]|       |-- 'wb' with value: '0'
[20:40:02]|       |-- 'awbg' with value: '0'
[20:40:02]|       |-- 'wbm' with value: 'Home'
[20:40:02]|       |-- 'ec' with value: '1'
[20:40:02]|       |-- 'aec2' with value: '1'
[20:40:02]|       |-- 'ael' with value: '0'
[20:40:02]|       |-- 'aecv' with value: '300'
[20:40:02]|       |-- 'gc' with value: '1'
[20:40:02]|       |-- 'agcg' with value: '0'
[20:40:02]|       |-- 'gcl' with value: '2X'
[20:40:02]|       |-- 'bpc' with value: '0'
[20:40:02]|       |-- 'wpc' with value: '1'
[20:40:02]|       |-- 'rg' with value: '1'
[20:40:02]|       |-- 'lenc' with value: '1'
[20:40:02]|       |-- 'hm' with value: '0'
[20:40:02]|       |-- 'vm' with value: '0'
[20:40:02]|       |-- 'dcw' with value: '1'
[... 78 lines cut ...]
[20:42:54]closing UDP socket
[20:42:54]closing UDP socket
[20:42:59]Creating TSP streamer
[20:42:59]Created streamer width=640, height=480
[20:42:59]Creating RTSP session
[20:43:00]RTSP received OPTIONS
[20:43:00]Creating TSP streamer
[20:43:00]Created streamer width=640, height=480
[20:43:00]Creating RTSP session
[20:43:01]RTSP received DESCRIBE
[20:43:02]RTSP received DESCRIBE
[20:43:03]RTSP received SETUP
[20:43:04]RTSP received TEARDOWN
[20:43:04]closing TCP socket
[20:43:04]closing UDP socket
[20:43:04]closing UDP socket
[20:43:05]RTSP received PLAY
```

Not sure what I'm doing wrong...

Can anyone help?

Best regards,
Andrew

rzeldent, 2025-03-15:
Is the camera url correct? rtsp://127.0.0.1:18554/camera.192_168_1_133? looks strange.
Take the URL from the Gui of the esp32cam-ready

andrewjsmith76, 2025-03-15:
Hi! Thanks for getting back to me :)

I think that's an artifact of how Home Assistant processes links... not sure but it may have something to do with the docker container it's hosted in??

The following link works in a browser:
http://192.168.1.221/stream

This one works in VLC (with quite some lag):
rtsp://192.168.1.221:554/mjpeg/1

If I try to add it into home assistant, I get these errors:
Logger: homeassistant.components.generic.config_flow
Source: components/generic/config_flow.py:203
integration: Generic Camera
Error getting camera image from http://192.168.1.221:/stream: TimeoutError

Logger: homeassistant.components.stream.stream.generic.test_stream
Source: components/stream/__init__.py:489
integration: Stream
Error from stream worker: Error opening stream (ERRORTYPE_5, I/O error, rtsp://192.168.1.221:554/mjpeg/1)

I guess this is a problem with home assistant.
RTSP streams from raspberry pis (e.g. using MediaMTX https://github.com/bluenviron/mediamtx?tab=readme-ov-file) work well with the Home Assistant integration.

Any thoughts?

andrewjsmith76, 2025-03-22:
Hi there... Still trying to make progress here... I'm thinking I can bypass Home Assistant and simply use a python script to read the images off the camera. If I enter `http://192.168.1.221/snapshot` in a browser, an image is displayed.

But using:
`curl http://192.168.1.221 -X GET:/snapshot `

Or

`curl http://192.168.1.221/snapshot`

in a Terminal yields:

`curl: (52) Empty reply from server`

What am I doing wrong?

Best regards,
Andrew

rzeldent, 2025-03-31:
Hi Andrew,

Might also indeed be something with the docker, container You need to specify the mapping of the network ports and RTSP has a communication port and an (dynamic) streaming port. This might be the cause...

andrewjsmith76, 2025-04-01:
Thank you for your comment... I'll look into it.

closed 2025-04-01 by andrewjsmith76

xthree, 2025-04-05:
> Thank you for your comment... I'll look into it.

were you able to fix this? I am having the same issue currently. I can view it in VLC just fine, but home assistant cant connect. I wonder if it has something to do with needing a https connection?

andrewjsmith76, 2025-04-06:
Sadly, no... 
It's still baffling to me since the RTSP streams from other cameras come through to HA just fine.