rzeldent/esp32cam-rtsp #164: jpeg not supported...how to change sensor type?

djepsylon, 2025-03-09:
hi,
i ve tested my esp32cam with the arduino ide exemple script and it work well with ai thinker as board and at line 72 i need to change for rgb565.
so, the exemple script is very limited and i try to use yours but i don't find where put the same sensor like in arduino ide script.
i have this error nellow in vscode, how can i solve this?
```
\-- [hidden]
04:35:45.208 > [  5701][V][main.cpp:228] initialize_camera(): initialize_camera
04:35:45.214 > [  5708][I][main.cpp:230] initialize_camera(): Frame size: HD (1280x720)
04:35:45.226 > [  5715][I][main.cpp:232] initialize_camera(): JPEG quality: 14
04:35:45.232 > [  5721][I][main.cpp:234] initialize_camera(): Frame duration: 10 ms
04:35:45.712 > E (6178) camera: JPEG format is not supported on this sensor
04:35:45.718 > [  6201][E][main.cpp:410] setup(): Failed to initialize camera. Error: 0x106. Frame size: HD (1280x720), frame rate: 10 ms, jpeg quality: 14
04:35:46.230 > [  6719][V][main.cpp:228] initialize_camera(): initialize_camera
04:35:46.236 > [  6725][I][main.cpp:230] initialize_camera(): Frame size: HD (1280x720)
04:35:46.241 > [  6731][I][main.cpp:232] initialize_camera(): JPEG quality: 14
04:35:46.246 > [  6737][I][main.cpp:234] initialize_camera(): Frame duration: 10 ms
04:35:46.728 > Camera probe failed with error 0x106E (7193) camera: JPEG format is not supported on this sensor
04:35:46.733 > [  7217][E][main.cpp:410] setup(): Failed to initialize camera. Error: 0x106. Frame size: HD (1280x720), frame rate: 10 ms, jpeg quality: 14
04:35:47.249 > [  7738][V][main.cpp:228] initialize_camera(): initialize_camera
04:35:47.255 > [  7744][I][main.cpp:230] initialize_camera(): Frame size: HD (1280x720)
04:35:47.260 > [  7750][I][main.cpp:232] initialize_camera(): JPEG quality: 14
04:35:47.265 > [  7756][I][main.cpp:234] initialize_camera(): Frame duration: 10 ms 
```

djepsylon, 2025-03-10:
i ve make some tries and i ve found how to make my rgb565 sensor in the script.

[image]

[image]

my sensor is correctly detected now but impossible to have picture, i have this error in the terminalk console:

[image]

rzeldent, 2025-03-15:
Hi Epsylon,

You came quite a long way; you managed to get RGB565 but unfortunately the streamer uses MJPEG. SO the frames are generated from the JPG data from the image.
I think it is possible to change this but you'll have to make a new streamer. On top of that, the amount of data will be much larger and doubt this can be handled by the limited processing power and WiFi bandwidth...

You can use the WebStreaming; this might work with maybe minimal changes for RGB 565.

closed 2026-09-04 by github-actions[bot]