rzeldent/esp32cam-rtsp #167: So which is the latest version?

JPMakesStuff, 2025-03-28:
Per the note at the top of the readme I used the develop branch, which seems ahead but results in the title of the device webpage saying "ESP32CAM-RTSP v1.0"

Then some users seem to be getting v1.6 which is maybe the result of the main branch, but unclear if it supports the perticular board I'm using "esp32cam_seeed_xiao_esp32s3_sense"

Documentation seems excellent, other than this confusion, or perhaps maybe it is just me that is confused?

rzeldent, 2025-03-31:
AppVersion still comes from the config.h so has never been updated; maybe this should have been done.
I think they're revering to the release version on github.

rzeldent, 2026-08-28:
The branch has just been merged to develop. So please also test it!

rzeldent, 2026-09-22:
Versions main and develop are the same and the latest version.

closed 2026-09-22 by rzeldent