rzeldent/esp32cam-rtsp #168: No connection to wiFi

joergsimon54, 2025-04-05:
Hello,

I try to connect to my local WiFi network but the cam does not connect. The cam works quite well wich the standard arduino example software. I use my ssid and password which works with other esp devices very well. 
I configured via Change settings the AP password, WiFi SSID and WiFi password, and use the Apply button.
Can you please help.

Best regards

Joerg

[image]
[image]
[image]
[image]
[image]

rzeldent, 2025-04-05:
Hi Simon,

Attach the serial console to see what's going on.

Also, remember the ESP32 only supports 2.4Ghz networks, not 5G ..

joergsimon54, 2025-04-05:
OK, now it Workshop, but dont know why.ThanksJoerg

closed 2026-09-04 by github-actions[bot]